DCC-EX/EX-Turntable #79: TTEX 0.3.1 Beta Test (alt.config, ULN2003_HALF_CCW) Results:

momfertje, 2022-05-24:
### Testing configuration

* CommandStation-EX version: 4.0.2 (add-turntable-controller version)
* Turntable-EX version: 0.3.1 Beta
* Microcontroller: Nano V3
* Stepper driver: ULN2003
* Stepper motor: 28BYJ-48
* Homing sensor: A3144
* Dual relay board: Active Low

`If you have edited config.h in Turntable-EX or myHalcpp or myAutomation.h in CommandStation-EX, please upload those to this issue also.`

### Test 1 - Startup Sequence

`PASS`

Any observations or comments:

### Test 2 - Turntable-EX is Available

`PASS`

Any observations or comments:

### Test 3 - Basic Movement with Automatic Phase Switching
#### Test 3, Part 1 - Clockwise

`PASS`

Any observations or comments:

#### Test 3, Part 2 - Counter Clockwise

`PASS`

Any observations or comments:

### Test 4 - Homing

`PASS`

Any observations or comments:

### Test 5 - Calibration

`PASS`

Any observations or comments:

### Test 6 - LED Output Testing

`PASS`

Any observations or comments:

### Test 7 - Accessory Output Testing

`PASS`

Any observations or comments:

### Test 8 - Manual Phase Switching

`PASS`

Any observations or comments:

### General comments or feedback

Please share any general comments or feedback here:

**Testing switch moment (calibrated at 4100 half step modus):**
`
Homing started
Turntable homed successfully
CALIBRATION: Phase 1, homing...
CALIBRATION: Phase 2, counting full turn steps...
CALIBRATION: Completed, storing full turn step count: 4093
Manual override has been set for 4093 steps per revolution
Automatic phase switching enabled at 45 degrees
Phase will switch at 495 steps from home, and revert at 2475 steps from home
Turntable homed successfully
Received notification to move to step postion 495
Position steps: 495, Auto phase switch - moving 1 steps
Setting phase switch flag to: 1
Received notification to move to step postion 2473
Position steps: 2473, Auto phase switch - moving 1978 steps
Setting phase switch flag to: 1
Received notification to move to step postion 2474
Position steps: 2474, Auto phase switch - moving 1 steps
Setting phase switch flag to: 1
Received notification to move to step postion 2475
Position steps: 2475, Auto phase switch - moving 1 steps
Setting phase switch flag to: 0
`
[config.h.txt](https://github.com/DCC-EX/Turntable-EX/files/8765867/config.h.txt)

peteGSX, 2022-05-25:
Hey @momfertje am I reading the output right that the manual step override was enabled, but that calibration happened automatically as well? That seems wrong.

momfertje, 2022-05-25:
NO, not the case.

I went using manual step override, to see what happened AFTER a manual initiated calibration. Works as might be expected:
- init calibration
- calibration executed with good end result
- afterwards the calibrated steps are used

This seems logical, because if someone is initiating calibration the purpose is to get a good working TT. So using the value from the calibration OVER the defined value makes sense.

May be an remark in the calibration section?

peteGSX, 2022-05-25:
Ah yes! I went through that but I didn't think to capture that in documentation. Will do so, good call!

peteGSX, 2022-05-25:
Add some clarification to the configuration options page: https://dcc-ex.com/turntable-ex/configure.html#full-step-count

closed 2022-05-25 by peteGSX