DCC-EX/EX-Turntable #80: TTEX 0.3.1 Beta Test  (ULN2003_FULL_CW) Results:

momfertje, 2022-05-24:
### Testing configuration

* CommandStation-EX version: 4.0.2 (add-turntable-controller version)
* Turntable-EX version: 0.3.1 Beta
* Microcontroller: Nano V3
* Stepper driver: ULN2003
* Stepper motor: 28BYJ-48
* Homing sensor: A3144
* Dual relay board: Active Low

`If you have edited config.h in Turntable-EX or myHalcpp or myAutomation.h in CommandStation-EX, please upload those to this issue also.`

### Test 1 - Startup Sequence

`PASS`

Any observations or comments:

### Test 2 - Turntable-EX is Available

`PASS`

Any observations or comments:

### Test 3 - Basic Movement with Automatic Phase Switching
#### Test 3, Part 1 - Clockwise

`PASS`

Any observations or comments:

#### Test 3, Part 2 - Counter Clockwise

`PASS`

Any observations or comments:

### Test 4 - Homing

`PASS`

Any observations or comments:

### Test 5 - Calibration

`PASS`

Any observations or comments:

### Test 6 - LED Output Testing

`PASS`

Any observations or comments:

### Test 7 - Accessory Output Testing

`PASS`

Any observations or comments:

### Test 8 - Manual Phase Switching

`PASS`

Any observations or comments:

### General comments or feedback

Please share any general comments or feedback here:

**Testing switch moment (Manual Override 4096 |FULL step modus):**

`Available at I2C address 0x60
Manual override has been set for 4096 steps per revolution
Automatic phase switching enabled at 45 degrees
Phase will switch at 495 steps from home, and revert at 2475 steps from home
Homing...
Homing started
Turntable homed successfully
Turntable homed successfully
CALIBRATION: Phase 1, homing...
CALIBRATION: Phase 2, counting full turn steps...
CALIBRATION: Completed, storing full turn step count: 2050
Manual override has been set for 2050 steps per revolution
Automatic phase switching enabled at 45 degrees
Phase will switch at 225 steps from home, and revert at 1125 steps from home
Turntable homed successfully
Received notification to move to step postion 224
Position steps: 224, Auto phase switch - moving 224 steps
Setting phase switch flag to: 0
Received notification to move to step postion 225
Position steps: 225, Auto phase switch - moving 1 steps
Setting phase switch flag to: 1
Received notification to move to step postion 1124
Position steps: 1124, Auto phase switch - moving 899 steps
Setting phase switch flag to: 1
Received notification to move to step postion 1125
Position steps: 1125, Auto phase switch - moving 1 steps
Setting phase switch flag to: 0
Homing started
Turntable homed successfully
`
[config.h.txt](https://github.com/DCC-EX/Turntable-EX/files/8766030/config.h.txt)

peteGSX, 2022-05-24:
Looks like results as expected, thanks Rico!

closed 2022-05-24 by peteGSX

momfertje, 2022-05-25:
You're welcome, Pete!
Looks like putting Serial Monitor output between back quotes is not a good idea :)